pyocd/pyOCD #1646: pyocd fails to write cortex_m0 / hk32f030mf4p6 chip after upgrade to 0.36

bmigette, 2023-11-25:
Hi,
Was trying to upgrade a controller with the following commands:
```
pyocd erase -c -t hk32f030mf4p6 --config ./pyocd.yaml
pyocd load ./Project/MDK5/Objects/LCM_Light_Control_IO_WS2812_New.hex -t hk32f030mf4p6 --config ./pyocd.yaml
```

The pack and config files above are in this repository:
https://github.com/surfdado/floatwheel/tree/main/LCM

The first erase went fine, however all subsequent erase and load command were failing with the following error:

```
0000875 I Erasing chip... [eraser]
0001000 W T bit in XPSR is invalid; the vector table may be invalid or corrupt [cortex_m]
0001032 C target was not halted as expected after calling flash algorithm routine (IPSR=3) [__main__]
```

After downgrading to 0.34.3, I was able to erase and load binary file

AmirrezaFiroozi, 2024-03-25:
I can confirm this issue

hagibr, 2024-07-29:
I'm with a similar issue. My workaround is to use 2 python environments, one with the latest (0.36.0) for some devices from ST and another one with the last version that worked with my specific device from Silabs (0.34.3).

mjakovlevs, 2024-09-03:
I have the same issue with STM32L051 target on pyOCD 0.36, this condition occurs after erase operation (`W T bit in XPSR is invalid; the vector table may be invalid or corrupt [cortex_m]`). Target works normally after erasing flash after downgrading to version 0.35.0.

closed 2025-08-13 by TeoMahnic