pyocd/pyOCD #1649: Adding one more picoprobe VID/PID

hagibr, 2023-12-01:
dmesg partial dump:

```
[ 5019.034338] usbcore: registered new interface driver cdc_acm
[ 5019.034344] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[ 5048.145303] usb 1-1.3: USB disconnect, device number 3
[ 5120.908559] usb 1-1.4: new full-speed USB device number 4 using xhci_hcd
[ 5121.017463] usb 1-1.4: New USB device found, idVendor=2e8a, idProduct=000c, bcdDevice= 1.01
[ 5121.017496] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5121.017513] usb 1-1.4: Product: Debug Probe (CMSIS-DAP)
[ 5121.017528] usb 1-1.4: Manufacturer: Raspberry Pi
[ 5121.017542] usb 1-1.4: SerialNumber: E6614103E73A3B2F
```

Solved by adding the line
```
SUBSYSTEM=="usb", ATTR{idVendor}=="2e8a", ATTR{idProduct}=="000C", MODE:="666"
```
to 50-picoprobe.rules

elfmimi, 2023-12-27:
The new picoprobe is based on CMSIS-DAP protocol. so I suppose it is natural to put it in 50-cmsis-dap.rules instead.

RobertRostohar, 2026-09-08:
Added with #2027.

closed 2026-09-08 by RobertRostohar