pyocd/pyOCD #1682: pyocd 0.36.0 version in python 3.12.2 can't find stlink v2 probe

3x00, 2024-03-29:
I test pyocd 0.36.0 in pytho 3.9.6 is works, it can find my stlink-v2 probe
pyocd 0.35.0 in python 3.11.0 is works too
when I install pyocd on 3.12.2 which pyocd version is 0.36.0, use <pyocd list> return <No available debug probes are connected>
the step of install is same

reed-dan, 2024-04-07:
@3x00 do you have a generic stlink-v2 probe, and if so what mcu is on the probe?

3x00, 2024-04-11:
> @3x00 do you have a generic stlink-v2 probe, and if so what mcu is on the probe?

I had try daplink and jlink, they can works.
python is 3.12.2 and pyocd is 0.36.0

elfmimi, 2024-04-14:
This is an issue caused by libusb-package not providing the _wheel_ for Python 3.12 .
If Python 3.12 is in use, pip will try to build `libusb-1.0.dll` .
For this build to complete successfully on the Windows platform,
VisualStudio must be properly installed on your local PC environment.
Otherwise pip will install libusb-package without `libusb-1.0.dll` .

The workaround is to manually get `libusb-1.0.dll` and place it where it should be.
https://github.com/libusb/libusb/releases/tag/v1.0.27
Put it to C:\Windows\System32\libusb-1.0.dll if you are not certain where to.

cf.
https://github.com/pyocd/libusb-package/issues/16
https://github.com/pyocd/libusb-package/pull/17

3x00, 2024-04-14:
well, python 3.12.3 is released, and I try it st-link is works now.

closed 2024-04-14 by 3x00

3x00, 2024-04-15:
> This is an issue caused by libusb-package not providing the _wheel_ for Python 3.12 . If Python 3.12 is in use, pip will try to build `libusb-1.0.dll` . For this build to complete successfully on the Windows platform, VisualStudio must be properly installed on your local PC environment. Otherwise pip will install libusb-package without `libusb-1.0.dll` .
> 
> The workaround is to manually get `libusb-1.0.dll` and place it where it should be. https://github.com/libusb/libusb/releases/tag/v1.0.27 Put it to C:\Windows\System32\libusb-1.0.dll if you are not certain where to.
> 
> cf. [pyocd/libusb-package#16](https://github.com/pyocd/libusb-package/issues/16) [pyocd/libusb-package#17](https://github.com/pyocd/libusb-package/pull/17)

you are right