pyocd/pyOCD #1692: TPIU not parsing and detect correcttly for STM32H7

fernicmarotta, 2024-05-06:
[image]

fernicmarotta, 2024-05-06:
When 0001441 I   [4]<e00f5000:TPIU CS-400 class=9 designer=43b:Arm part=912 devtype=1
1 archid=0000 devid=a0:0:0> [rom_table]
The parse         tpiu = self._target.get_first_child_of_type(TPIU) result false
0001480 W SWV not initalized: Target does not have TPIU component [swv]

rmolinete, 2025-01-30:
I also have the same issue. 
Using a Nucleo H743ZI2 board (in-board ST-Link V3) I can not have SWV. 
```
0000692 I   [4]<e00f5000:TPIU CS-400 class=9 designer=43b:Arm part=912 devtype=11 archid=0000 devid=a0:0:0> [rom_table]
0000696 I CPU core #0: Cortex-M7 r1p1, v7.0-M architecture [cortex_m]
0000696 I   Extensions: [DSP, FPU, FPU_DP, FPU_V5, MPU] [cortex_m]
0000696 I   FPU present: FPv5-D16-M [cortex_m]
0000697 I 4 hardware watchpoints [dwt]
0000700 I 8 hardware breakpoints, 1 literal comparators [fpb]
0000711 I Semihosting will be output to console [gdbserver]
0000711 W SWV not initalized: Target does not have TPIU component [swv]
```

If I use SWV from STM32CubeProgrammer tool I can see the prints though.