pyocd/pyOCD #1694: No Connected Debug Probes when running executable built with PyInstaller

Dominikisses, 2024-05-15:
I am using pyocd version 0.36 with Python 3.11 and have used PyInstaller to build an executable of a script that uses pyocd. My script includes the use of ConnectHelper to open a session with an STM32 board. When I run the script (.py file), the node is found and everything works correctly. However, when I run the built executable, I encounter the following error:

←[31mNo connected debug probes←[0m

Steps to Reproduce:

1.     Write a Python script that uses pyocd to connect to an STM32 board.
2.     Use PyInstaller to build an executable from this script.
3.     Run the executable.

Expected Behavior:

The executable should successfully connect to the STM32 board, just as the script does when run directly with Python.

Actual Behavior:

The executable fails to find any connected debug probes and outputs the following error message:

←[31mNo connected debug probes←[0m

Additional Information:

- Python Version: 3.11
- pyocd Version: 0.36
- PyInstaller Command Used: pyinstaller --onefile my_script.py
- Operating System: [Windows 10]

Questions:

- Is there a known issue with pyocd and PyInstaller?
- Are there specific steps or configurations needed to ensure pyocd works in a PyInstaller-built executable?
- Could there be a missing dependency or path issue in the executable that isn't present when running the script directly?

Any help or guidance on resolving this issue would be greatly appreciated.

Thank you!

julian-0, 2024-05-29:
Hi!

Are you running the .exe in the same computer where the .py file works or is in another device?

I think i had this error when I generated the exe and tried to use it in another device without installing the [st-link driver](https://www.st.com/en/development-tools/stsw-link009.html).

Also in this [issue ](https://github.com/pyocd/pyOCD/issues/1529)whe discussed about the exe generation, maybe it helps.

FangYuFan, 2024-07-22:
same issue like you. do you find a way to fix it?

Dominikisses, 2024-07-23:
> same issue like you. do you find a way to fix it?

Try to change the .spec file which is generated while you try to generate the .exe with pyinstaller.

it works for my .exe with following .spec file:

`# -*- mode: python ; coding: utf-8 -*-

import os
import pathlib

from PyInstaller.utils.hooks import get_package_paths, collect_entry_point

datas_probe, hiddenimports_probe = collect_entry_point('pyocd.probe')
datas_rtos, hiddenimports_rtos = collect_entry_point('pyocd.rtos')
datas = [
    (get_package_paths('pyocd')[1], 'pyocd'),
    (get_package_paths('pylink')[1], 'pylink')
]

# Right now we exclude cmsis_pack_manager, but could be needed in the future
# datas.append((get_package_paths('cmsis_pack_manager')[1], 'cmsis_pack_manager'))
excludes = ['cmsis_pack_manager']

# For the CLI version we don't want to include the gui command
excludes.append('tkinter')

a = Analysis(
    ['main.py'],
    pathex=['../'],
    binaries=[],
    datas= datas + datas_probe + datas_rtos,
    hiddenimports=hiddenimports_probe + hiddenimports_rtos,
    hookspath=[],
    hooksconfig={},
    runtime_hooks=[],
    excludes=excludes,
    win_no_prefer_redirects=False,
    win_private_assemblies=False,
    cipher=None,
    noarchive=False
)
pyz = PYZ(a.pure,
          a.zipped_data,
          cipher=None)

exe = EXE(
    pyz,
    a.scripts,
    a.binaries,
    a.datas,
    [],
    name='NeoMonitor_KM',
    debug=False,
    bootloader_ignore_signals=False,
    strip=False,
    upx=False,
    upx_exclude=[],
    runtime_tmpdir=None,
    console=True,
    disable_windowed_traceback=False,
    argv_emulation=False,
    target_arch=None,
    codesign_identity=None,
    entitlements_file=None,
    icon=['logo.ico'],
)
`

then you have to create the .exe with:
`pyinstaller options name.spec`

you can check it out at [](https://pyinstaller.org/en/stable/spec-files.html)

FangYuFan, 2024-07-24:
> > same issue like you. do you find a way to fix it?
> 
> Try to change the .spec file which is generated while you try to generate the .exe with pyinstaller.
> 
> it works for my .exe with following .spec file:
> 
> ```python
> # -*- mode: python ; coding: utf-8 -*-
> 
> # Import necessary modules
> import os
> import pathlib
> 
> # Import specific utilities from PyInstaller
> from PyInstaller.utils.hooks import get_package_paths, collect_entry_point
> 
> # Collect data and hidden imports for pyocd.probe and pyocd.rtos
> datas_probe, hiddenimports_probe = collect_entry_point('pyocd.probe')
> datas_rtos, hiddenimports_rtos = collect_entry_point('pyocd.rtos')
> 
> # Define the data paths for pyocd and pylink
> datas = [
>     (get_package_paths('pyocd')[1], 'pyocd'),
>     (get_package_paths('pylink')[1], 'pylink')
> ]
> 
> # Right now we exclude cmsis_pack_manager, but could be needed in the future
> # datas.append((get_package_paths('cmsis_pack_manager')[1], 'cmsis_pack_manager'))
> excludes = ['cmsis_pack_manager']
> 
> # For the CLI version we don't want to include the gui command
> excludes.append('tkinter')
> 
> # Define the analysis parameters
> a = Analysis(
>     ['main.py'],
>     pathex=['../'],
>     binaries=[],
>     datas= datas + datas_probe + datas_rtos,
>     hiddenimports=hiddenimports_probe + hiddenimports_rtos,
>     hookspath=[],
>     hooksconfig={},
>     runtime_hooks=[],
>     excludes=excludes,
>     win_no_prefer_redirects=False,
>     win_private_assemblies=False,
>     cipher=None,
>     noarchive=False
> )
> 
> # Define the PYZ parameters
> pyz = PYZ(a.pure,
>           a.zipped_data,
>           cipher=None)
> 
> # Define the EXE parameters
> exe = EXE(
>     pyz,
>     a.scripts,
>     a.binaries,
>     a.datas,
>     [],
>     name='program_name',
>     debug=False,
>     bootloader_ignore_signals=False,
>     strip=False,
>     upx=False,
>     upx_exclude=[],
>     runtime_tmpdir=None,
>     console=True,
>     disable_windowed_traceback=False,
>     argv_emulation=False,
>     target_arch=None,
>     codesign_identity=None,
>     entitlements_file=None,
>     icon=['logo.ico'],
> )
> ```
> 
> then you have to create the .exe with: `pyinstaller options name.spec`
> 
> you can check it out at [pyinstaller_doc](https://pyinstaller.org/en/stable/spec-files.html)

thanks, it works for me.

closed 2025-08-08 by TeoMahnic