pyocd/pyOCD #1696: STM32H753 binary fails to run

Dillonmcewan, 2024-05-20:
I'm attempting to flash a very basic LED blinky app onto a NUCLEO-H753ZI. When I flash the board with pyOCD I see output indicating the flash was successful: `Erased 131072 bytes (1 sector), programmed 54272 bytes (53 pages), skipped 0 bytes (0 pages)`. But the app doesn't appear to start even after issuing a reset command. I started stepping through the code and it appears to hit a hard fault somewhere during RCC init. The weird thing is that if I flash the same binary using [probe-rs](https://probe.rs/) it runs without any issue, which makes me believe there's some problem in the flashing process itself rather than the binary.

nattgris, 2024-06-12:
You should be able to use the builtin target `stm32h743xx` which is the same chip but without crypto peripherals. I've found its flash loader to be working, compared to the CMSIS pack version, which corrupts the flash with high probability.

However note my https://github.com/pyocd/pyOCD/issues/1700#issuecomment-2162965594 that I think the sector sizes are specified incorrectly for that target. I'll send a PR with a fix, perhaps you can assist in testing it on a H743?

Dillonmcewan, 2024-06-12:
Thanks @nattgris, using the built-in target seems to work. I'd be happy to help with testing if you link the PR.

nattgris, 2024-06-13:
Great, #1703 is changing the sector sizes and should (still) work for your chip. Not sure if the incorrect sector size actually posed a problem, could depend on whether doing chip or sector erase.

I also noticed by just using that H743 target unmodified on my H750, that it was constantly reported by pyocd commander to be in security locked state, due to a bug in parsing the RDP value. Is that something you noticed with the builtin target? It should be fixed in my PR.

Dillonmcewan, 2024-06-19:
Got some time to test this today. I was able to flash my H753 with the `stm32h750xx` target several times without any issues.

I also checked the h743 target and did not see the same issue with getting stuck in a security locked state