pyocd/pyOCD #1698: Fails to flash/erase NXP MIMXRT1170-EVK with updated MCUXpresso SDK

DerekSnell, 2024-05-31:
The NXP MCUXpresso SDK v2.15 updated the Flash Control Block (FCB) used for the MIMXRT1170-EVK board .  These changes to the flash boot settings have broken pyOCD support for this board now.  If an app built with the new FCB is downloaded to the board, pyOCD succeeds in programming that app.  But after booting that app, pyOCD is no longer able to erase or program the flash.  This likely also affects the MIMXRT1160-EVK board, although I have not tested it.  This issue was originally reported at https://github.com/zephyrproject-rtos/zephyr/issues/73234.  The attached [zephyr.hex](https://github.com/user-attachments/files/15516064/zephyr.zip) is an app showing this issue.

To replicate, build an app using the v2.15 SDK or later.  Zephyr apps with this issue use commit  [e4e463a](https://github.com/zephyrproject-rtos/zephyr/commit/e4e463af812e61bb5bfb4b1087d48e1dbcd9d8d5) or later.  Program the flash using a command like:
```
pyocd flash -e sector -a 0x30000000 -t mimxrt1170_cm7 zephyr.hex
```
The initial programming succeeds.  Reset the board.  Try the same programming command, and it fails.  Also, the erase command like this fails:
```
pyocd erase -t mimxrt1170_cm7 --mass
```
To recover the board, I can use another tool to erase or reprogram the flash.  LinkServer is NXP's tool that works with the same CMSIS-DAP debug probes.  This is the LinkServer command to erase the flash:
```
/usr/local/LinkServer/LinkServer flash MIMXRT1176xxxxx:MIMXRT1170-EVK erase
```
Or another option, change the DIP switch settings on the EVK to boot in serial downloader mode.  Use NXP's bootloader tools to erase the flash.

Thanks

ithinuel, 2024-06-04:
Hi, thank you for reporting this issue.

If I understand correctly, what would need to be updated/fixed is the binary blob of the Flash Algorithms in https://github.com/pyocd/pyOCD/blob/main/pyocd/target/builtin/target_MIMXRT1176xxxxx.py#L42-L775

I don’t know where the source for that blob lives and if there’s an updated version that’d be backward compatible (the flash algo needs to be able to flash firmware with either of the current and the previous FCB format).

Would someone at NXP be able to help provide an update for that?

DerekSnell, 2024-06-05:
Hi @ithinuel ,
NXP provides the [LinkServer](https://www.nxp.com/design/design-center/software/development-software/mcuxpresso-software-and-tools-/linkserver-for-microcontrollers:LINKERSERVER) tool which supports CMSIS-DAP debug probes.  NXP's tool support is focused on LinkServer, and I am not aware of NXP supporting pyOCD.  I am not too familiar with how the flash algorithms are managed in these tools, but perhaps the algorithm used in LinkServer can be leveraged by pyOCD.

Best regards

ithinuel, 2024-06-05:
@dleach02 would you know someone familiar with NXP’s ecosystem and with some bandwidth who’d be able help on this matter?

dleach02, 2024-06-05:
@ithinuel I did check to see if we actually have someone that supports pyOCD. Appears we do not and have not actively supported it for a while. The last person to touch this no longer works for us. I'm not sure though how important this really is for us because we have two different solutions for support between LinkServer and Segger.

IvoBCD, 2024-06-06:
@dleach02 Unfortunately both are closed-source and LinkServer is not available for Linux/Aarch64 on which we'd like to run SPSDK, so pyocd would have been useful.