pyocd/pyOCD #1713: Error during board uninit and no ACK after that

vacagonzalo2, 2024-07-29:
# Intended use

I want to flash and erase my target (multiple times if needed) from a Raspberry Pi. I have to use a raw binary in this case, I cannot use the HEX format.

# Result

I was able to program the target and its working. However I can only erase or re program it from a vendor's IDE (MPLAB X).

> Note: with the HEX format (Intel format) everything works great.

# System information

Debug probe: ATMEL-ICE

HOST: Raspberry Pi 4B

```
ifupper@raspberrypi
-------------------
OS: Debian GNU/Linux 12 (bookworm) aarch64
Host: Raspberry Pi 4 Model B Rev 1.5
Kernel: 6.6.31+rpt-rpi-v8
Uptime: 4 hours, 12 mins
Packages: 752 (dpkg)
Shell: bash 5.2.15
Terminal: /dev/pts/1
CPU: (4) @ 1.800GHz
Memory: 825MiB / 3792MiB
```

```
ifupper@raspberrypi:/tmp $ pip list
Package             Version
------------------- -----------
appdirs             1.4.4
asgiref             3.6.0
Babel               2.10.3
blinker             1.5
capstone            4.0.2
certifi             2022.9.24
cffi                1.16.0
chardet             5.1.0
charset-normalizer  3.0.1
click               8.1.3
cmsis-pack-manager  0.5.3
colorama            0.4.6
colorzero           2.0
connector           0.1
cryptography        38.0.4
distro              1.8.0
Flask               2.2.2
future              0.18.2
gpiozero            2.0
idna                3.3
importlib_metadata  8.2.0
importlib_resources 6.4.0
intelhex            2.3.0
intervaltree        3.0.2
itsdangerous        2.1.2
Jinja2              3.1.2
lark                1.1.9
lgpio               0.2.2.0
libusb-package      1.0.26.2
MarkupSafe          2.1.2
natsort             8.4.0
paho-mqtt           1.6.1
pigpio              1.78
pip                 23.0.1
prettytable         3.10.2
psutil              6.0.0
pycparser           2.22
pycryptodome        3.20.0
pycryptodomex       3.11.0
pyelftools          0.29
pyinotify           0.9.6
pylink-square       1.2.1
pymodbus            3.0.0rc1
pyocd               0.36.0
pyOpenSSL           23.0.0
pyserial            3.5
pyserial-asyncio    0.6
python-apt          2.6.0
python-dotenv       0.21.0
pytz                2022.7.1
pyusb               1.2.1.post2
PyYAML              6.0
ranger-fm           1.9.3
requests            2.28.1
RPi.GPIO            0.7.1a4
setuptools          66.1.1
simplejson          3.18.3
six                 1.16.0
smbus2              0.4.2
sortedcontainers    2.4.0
spidev              3.5
ssh-import-id       5.10
toml                0.10.2
typer               0.7.0
typing_extensions   4.12.2
ufw                 0.36.2
urllib3             1.26.12
wcwidth             0.2.13
websocket-client    1.2.3
Werkzeug            2.2.2
wheel               0.38.4
zipp                3.19.2
```

# Log

```
ifupper@raspberrypi:/tmp $ pyocd load firmware.bin --target atsamc20e18a
0002149 I Loading /tmp/firmware.bin [load_cmd]
[==================================================] 100%
0027060 I Erased 32768 bytes (2 sectors), programmed 32768 bytes (32 pages), skipped 229376 bytes (224 pages) at 10.28 kB/s [loader]
0027080 E Error during board uninit: [session]
ifupper@raspberrypi:/tmp $ pyocd commander --target atsamc20e18a
0002558 E Error while initing target: No ACK received [commander]
ifupper@raspberrypi:/tmp $ pyocd erase --mass --target atsamc20e18a
0002062 C No ACK received [__main__]
```

hagibr, 2024-07-29:
Hello, I never worked with ATSAMC20 series, but I think you need the cmsis pack file for this family to make things work. You can download it at https://packs.download.microchip.com/, or directly by this link: https://packs.download.microchip.com/Microchip.SAMC20_DFP.3.5.93.atpack
Then, add this argument to your command line: ```--pack <path-to-pack-file>```

vacagonzalo2, 2024-07-29:
@hagibr thank you for your response.
Yes, I had installed the pack with the command: `pyocd pack install atsamc20e18a`

If I run `pyocd pack show`, this is what I get:

```
ifupper@raspberrypi:~ $ pyocd pack show
  Pack             Version
----------------------------
  Keil.SAM-C_DFP   1.4.3
```

The problem seems to be that I cannot flash the target more than once. I have to erase it from the IDE.

This problem is also present on Windows 10 pro.

vacagonzalo2, 2024-07-29:
more info on the packs...
on the raspberrypi:

```
ifupper@raspberrypi:~ $ pyocd pack find atsamc20
  Part            Vendor      Pack             Version   Installed
--------------------------------------------------------------------
  ATSAMC20E15A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20E16A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20E17A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20E18A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20G15A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20G16A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20G17A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20G18A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20J15A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20J16A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20J17A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20J17AU   Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20J18A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20J18AU   Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20N17A    Microchip   Keil.SAM-C_DFP   1.4.3     True
  ATSAMC20N18A    Microchip   Keil.SAM-C_DFP   1.4.3     True
```

on Windows:

```
F:\gnvaca>pyocd pack find samc20
  Part            Vendor      Pack                   Version   Installed
--------------------------------------------------------------------------
  ATSAMC20E15A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20E16A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20E17A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20E18A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20G15A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20G16A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20G17A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20G18A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20J15A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20J16A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20J17A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20J17AU   Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20J18A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20J18AU   Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20N17A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
  ATSAMC20N18A    Microchip   Microchip.SAMC20_DFP   3.5.93    True
```

vacagonzalo2, 2024-08-02:
I was able to avoid this problem by excecuting `pyocd erase --mass --target atsamc20e18a` before loading the binary.

closed 2024-08-02 by vacagonzalo2