pyocd/pyOCD #1746: About pyocd updating.

lennvn, 2024-12-03:
May I ask is there anybody else handling the pull requests? And which branch should the PR merged into, main or develop?

mathias-arm, 2024-12-04:
The branch depends on the content of the PR. See https://github.com/pyocd/pyOCD/blob/main/docs/developers_guide.md#branch-configuration-policy

lennvn, 2024-12-05:
Thanks @mathias-arm
There's one more question, how to solve this problem on branch 'develop'?
[image]

mathias-arm, 2024-12-05:
Without the link to the PR or GitHub Actions run, it's impossible to help.

lennvn, 2024-12-06:
Here is my PR link. 
https://github.com/pyocd/pyOCD/pull/1747
Thanks.

mathias-arm, 2024-12-06:
I think `develop` is missing this change https://github.com/pyocd/pyOCD/commit/5166025ae5da5e093d6cfe2b26cae5e1334476e4

lennvn, 2024-12-06:
Thank you so much@mathias-arm. The problem seems to have been solved.

closed 2024-12-06 by lennvn