pyocd/pyOCD #1751: Most ST MCUs are missing

tdjastrzebski, 2025-01-13:
It seems that most ST's MCUs are missing. Only STM32L1 and STM32F3 MCUs are available in index.
What is going on?

```
C:\Windows\System32>pyocd --version
0.36.0

C:\Windows\System32>pyocd pack clean
0000743 I Removing all pack data... [pack_cmd]

C:\Windows\System32>pyocd pack update
0000549 I Updating pack index... [pack_cmd]
Downloading descriptors (1584/1584)

C:\Windows\System32>pyocd pack find stm32u5
0002908 W No matching devices. Please make sure the pack index is up to date. [pack_cmd]

C:\Windows\System32>pyocd pack find stm32g0
0000726 W No matching devices. Please make sure the pack index is up to date. [pack_cmd]

C:\Windows\System32>pyocd pack find stm32l0
0000660 W No matching devices. Please make sure the pack index is up to date. [pack_cmd]

C:\Windows\System32>pyocd pack find stm32l1
  Part             Vendor               Pack                 Version   Installed
----------------------------------------------------------------------------------
  STM32L100C6Ux    STMicroelectronics   Keil.STM32L1xx_DFP   1.4.1     False
  STM32L100C6UxA   STMicroelectronics   Keil.STM32L1xx_DFP   1.4.1     False
  STM32L100R8Tx    STMicroelectronics   Keil.STM32L1xx_DFP   1.4.1     False
  STM32L100R8TxA   STMicroelectronics   Keil.STM32L1xx_DFP   1.4.1     False
  STM32L100RBTx    STMicroelectronics   Keil.STM32L1xx_DFP   1.4.1     False
  STM32L100RBTxA   STMicroelectronics   Keil.STM32L1xx_DFP   1.4.1     False
  STM32L100RCTx    STMicroelectronics   Keil.STM32L1xx_DFP   1.4.1     False
  STM32L151C6Tx    STMicroelectronics   Keil.STM32L1xx_DFP   1.4.1     False
  STM32L151C6TxA   STMicroelectronics   Keil.STM32L1xx_DFP   1.4.1     False
  STM32L151C6Ux    STMicroelectronics   Keil.STM32L1xx_DFP   1.4.1     False
  STM32L151C6UxA   STMicroelectronics   Keil.STM32L1xx_DFP   1.4.1     False
...
```

grifcj, 2025-02-14:
I am seeing the same thing. 

I'm using pyocd 0.36.0 and cmsis-pack-manager 0.5.3.

I can't install packs on my Windows 10 and 11 machines. It did work on my linux machine, but took a while. 

I understand that pyocd defers to cmsis-pack-manager for handling cmsis packages. If I use run the command 'pack-manager print-cache-dir', the returned directory doesn't exist on Windows. On linux, it is full of *.pdsc files.

elfmimi, 2025-02-15:
It appears that `pyocd pack update` does not consistently cache all available packs. Some packs may be missing randomly.
Probably due to a server-side rate limit.

I could reproduce both successful complete update and successful but incomplete update, as of today.

You can manually complement missing packs. See this comment for details.
https://github.com/pyocd/pyOCD/issues/1602#issuecomment-1871118285