pyocd/pyOCD #1763: Installation on macOS (M1) fails

gregorFeigel, 2025-03-01:
I tried to install `pyOCD` as part of the `Zephyr` SDK on my Mac Mini M1. However, I'm facing an issues:

No matter if using the zephyr installer command `west packages pip --install` or use `pip install pyocd`, ` pip install --no-binary capstone pyocd` or `python3 -mpip install -U pyocd`, the installation always fails with: 

```bash
Using cached pycparser-2.22-py3-none-any.whl (117 kB)
Building wheels for collected packages: capstone
  Building wheel for capstone (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [71 lines of output]
      running bdist_wheel
      running build
# ....
 arch/M680X/M680XDisassembler.c:2124:20: warning: variable 's_cpu_type' is not needed and will not be emitted [-Wunneeded-internal-declaration]
       2124 | static const char *s_cpu_type[] = {
            |                    ^~~~~~~~~~
        CC      arch/EVM/EVMInstPrinter.o
        CC      arch/EVM/EVMMapping.o
        CC      arch/EVM/EVMModule.o
        CC      MCInst.o
        GEN     capstone.pc
      1 warning generated.
        LINK    libcapstone.so.4
        AR      libcapstone.a
      ar: creating archive ./libcapstone.a
      ld: unknown options: -soname
      clang: error: linker command failed with exit code 1 (use -v to see invocation)
      make: *** [libcapstone.so.4] Error 1
      error: [Errno 2] No such file or directory: 'libcapstone.4dylib'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for capstone
  Running setup.py clean for capstone
Failed to build capstone
ERROR: Failed to build installable wheels for some pyproject.toml based projects (capstone)
```
or 
```bash
# ... 
Building wheels for collected packages: capstone
  Building wheel for capstone (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [72 lines of output]
      running bdist_wheel
      running build
      Building C extensions
# ..
      error: [Errno 2] No such file or directory: 'libcapstone.4.dylib'
      [end of output]
  
  note: This error originates from a subprocess and is likely not a problem with pip.
  ERROR: Failed building wheel for capstone
  Running setup.py clean for capstone
Failed to build capstone
ERROR: Failed to build installable wheels for some pyproject.toml-based projects (capstone)
Traceback (most recent call last):
  File "/Users/REDACTED/zephyrproject/.venv/lib/python3.13/site-packages/west/app/main.py", line 561, in run_command
    self.run_extension(args.command, argv)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
  File "/Users/REDACTED/zephyrproject/.venv/lib/python3.13/site-packages/west/app/main.py", line 716, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
    ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                 config=self.config)
                 ^^^^^^^^^^^^^^^^^^^
# ... (much more)
```

Would be happy if someone can help me out, thanks.

chrisyunker, 2025-04-29:
I'm having the same problem. Steps to reproduce:

- Clone the `pyOCD` repo locally
- Install fresh version of python 3.13.3 via pyenv
- Set the python version of the `pyocd/pyOCD` directory to this version: `pyenv local 3.13.3`
- Attempt to install `pyOCD` from local repo: `pip install .`

I get the following output:
```
[10:25:22|~/git/github.com/pyocd/pyOCD] [main ?]
$ pip install .
Processing /Users/cyunker/git/github.com/pyocd/pyOCD
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting capstone<5.0,>=4.0 (from pyocd==0.36.1.dev9)
  Using cached capstone-4.0.2.tar.gz (2.0 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done

(snip)

Using cached pycparser-2.22-py3-none-any.whl (117 kB)
Building wheels for collected packages: pyocd, capstone
  Building wheel for pyocd (pyproject.toml) ... done
  Created wheel for pyocd: filename=pyocd-0.36.1.dev9-py3-none-any.whl size=14106799 sha256=368db3db00172ebdc883b0b2bdec481d941654c107be047ce5408e5542579269
  Stored in directory: /private/var/folders/bd/l_7hg1rx11xdzkxj973h9xh00000gn/T/pip-ephem-wheel-cache-ov9avf0t/wheels/34/d6/35/a700e6759bf076994d0b9484f4d6d17e9b6b0bad1792827a5b
  Building wheel for capstone (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for capstone (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [85 lines of output]
      /private/var/folders/bd/l_7hg1rx11xdzkxj973h9xh00000gn/T/pip-build-env-zrefvlhp/overlay/lib/python3.13/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
      !!

              ********************************************************************************
              Please consider removing the following classifiers in favor of a SPDX license expression:

              License :: OSI Approved :: BSD License

              See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
              ********************************************************************************

      !!
        self._finalize_license_expression()
      running bdist_wheel
      running build
      Building C extensions
        CC      cs.o
        CC      utils.o
        CC      SStream.o
        CC      MCInstrDesc.o
        CC      MCRegisterInfo.o
        CC      arch/ARM/ARMDisassembler.o
        CC      arch/ARM/ARMInstPrinter.o
        CC      arch/ARM/ARMMapping.o
        CC      arch/ARM/ARMModule.o
        CC      arch/AArch64/AArch64BaseInfo.o
        CC      arch/AArch64/AArch64Disassembler.o
        CC      arch/AArch64/AArch64InstPrinter.o
        CC      arch/AArch64/AArch64Mapping.o
        CC      arch/AArch64/AArch64Module.o
        CC      arch/M68K/M68KDisassembler.o
        CC      arch/M68K/M68KInstPrinter.o
        CC      arch/M68K/M68KModule.o
        CC      arch/Mips/MipsDisassembler.o
        CC      arch/Mips/MipsInstPrinter.o
        CC      arch/Mips/MipsMapping.o
        CC      arch/Mips/MipsModule.o
        CC      arch/PowerPC/PPCDisassembler.o
        CC      arch/PowerPC/PPCInstPrinter.o
        CC      arch/PowerPC/PPCMapping.o
        CC      arch/PowerPC/PPCModule.o
        CC      arch/Sparc/SparcDisassembler.o
        CC      arch/Sparc/SparcInstPrinter.o
        CC      arch/Sparc/SparcMapping.o
        CC      arch/Sparc/SparcModule.o
        CC      arch/SystemZ/SystemZDisassembler.o
        CC      arch/SystemZ/SystemZInstPrinter.o
        CC      arch/SystemZ/SystemZMapping.o
        CC      arch/SystemZ/SystemZMCTargetDesc.o
        CC      arch/SystemZ/SystemZModule.o
        CC      arch/X86/X86DisassemblerDecoder.o
        CC      arch/X86/X86Disassembler.o
        CC      arch/X86/X86IntelInstPrinter.o
        CC      arch/X86/X86ATTInstPrinter.o
        CC      arch/X86/X86Mapping.o
        CC      arch/X86/X86Module.o
        CC      arch/XCore/XCoreDisassembler.o
        CC      arch/XCore/XCoreInstPrinter.o
        CC      arch/XCore/XCoreMapping.o
        CC      arch/XCore/XCoreModule.o
        CC      arch/TMS320C64x/TMS320C64xDisassembler.o
        CC      arch/TMS320C64x/TMS320C64xInstPrinter.o
        CC      arch/TMS320C64x/TMS320C64xMapping.o
        CC      arch/TMS320C64x/TMS320C64xModule.o
        CC      arch/M680X/M680XDisassembler.o
        CC      arch/M680X/M680XInstPrinter.o
        CC      arch/M680X/M680XModule.o
        CC      arch/EVM/EVMDisassembler.o
      arch/M680X/M680XDisassembler.c:2124:20: warning: variable 's_cpu_type' is not needed and will not be emitted [-Wunneeded-internal-declaration]
       2124 | static const char *s_cpu_type[] = {
            |                    ^~~~~~~~~~
        CC      arch/EVM/EVMInstPrinter.o
        CC      arch/EVM/EVMMapping.o
      1 warning generated.
        CC      arch/EVM/EVMModule.o
        CC      MCInst.o
        GEN     capstone.pc
        LINK    libcapstone.so.4
        AR      libcapstone.a
      ar: creating archive ./libcapstone.a
      ld: unknown options: -soname
      clang: error: linker command failed with exit code 1 (use -v to see invocation)
      make: *** [libcapstone.so.4] Error 1
      make: *** Waiting for unfinished jobs....
      error: [Errno 2] No such file or directory: 'libcapstone.4.dylib'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for capstone
Successfully built pyocd
Failed to build capstone

[notice] A new release of pip is available: 25.0.1 -> 25.1
[notice] To update, run: pip install --upgrade pip
ERROR: Failed to build installable wheels for some pyproject.toml based projects (capstone)
```

chrisyunker, 2025-05-12:
I figured out the issue with my setup. I don't know the exact issue, but it had something to do with the macOS native Clang/LLVM tools conflicting with Homebrew installed LLVM tools. I had the following environment variables set up to use the Homebrew version, and once I removed them it fixed the problem:
```
#  export PATH="$HOMEBREW_PREFIX/opt/llvm/bin:$PATH"
#  export LDFLAGS="-L$HOMEBREW_PREFIX/opt/llvm/lib"
#  export CPPFLAGS="-I$HOMEBREW_PREFIX/opt/llvm/include"
#  export CFLAGS="-I$HOMEBREW_PREFIX/opt/llvm/include"
#  export CC="$HOMEBREW_PREFIX/opt/llvm/bin/clang"
#  export CXX="$HOMEBREW_PREFIX/opt/llvm/bin/clang++"
```